TheGammaSqueeze/GammaOS #107: Update gpSP

jloh, 2023-12-21:
First off, thanks a bunch for GammaOS! It runs great on my RG405M.

I'm using the gpSP emulator for some Pokemon games I'm playing and currently the poison effect outside of battle doesn't work because the bundled version of gpSP doesn't have it implemented.

I raised an issue on their GitHub (https://github.com/libretro/gpsp/issues/232) and they've said updating to a more recent release should fix it.

Is it currently possible to do that or does it require a new release of GammaOS?

TheGammaSqueeze, 2023-12-21:
Update as you would with any RetroArch core

https://github.com/TheGammaSqueeze/GammaOS/assets/116582950/ec3d6abd-6e87-4e87-8691-85ea36faf7c4

closed 2023-12-21 by TheGammaSqueeze

jloh, 2023-12-21:
Thanks so much! Didn’t know that was a thing.