TheGammaSqueeze/GammaOS #112: Upgrade gammaOS lite application

poc-sm, 2024-01-04:
Hello
Today we depend to @TheGammaSqueeze for the upgrade of various app like in GammaOS Lite
* retroarch 1.15.0 on GammaOS but 1.16.0 is available
* Daijishou 1.4.62 but 1.4.67 is available
* magisk (but not necessary)

I thing this is only an issue with the lite version 
GammaOS with google service, app upgrade may be automatic ? if someone can confirm

So I thinking about for the next build 
install f-droid and install retroarch from here 
It's not possible for now with daijishou 
https://github.com/TapiocaFox/Daijishou/issues/513
(may be just install the apk ?)

The main purpose is to avoid #64 for exemple and keep retroarch up to date

TheGammaSqueeze, 2024-02-23:
Usually I would expect the user to be able to download and install their own APK for updating stuff on GammaOS Lite.
With that being said, I have no objection with bundling something like a pre-set Obtainium https://youtu.be/9ULGZM9d5YM?t=254

poc-sm, 2024-02-23:
I give it a try, I didn't know about this kind off app but seems very useful to upgrade apk from github automatically (f-droid is already auto upgrade so not really usefull) 
It can be a good idea for a futur release.

poc-sm, 2024-02-26:
Hello 
I just give a try this week end (tested with ppsspp) and it's really good 
Obtainium can be downloaded from the f-droid and can be updated this way too https://f-droid.org/en/packages/dev.imranr.obtainium.fdroid/ 

Do you think it's a good idea to privide f-droid by default with gammaos lite and obtainium configured to check what's it's not on fdroid ?

david79111, 2024-06-10:
you can use aurora store (same apps as play store but make sure to login with google acc) or obtanium for emulators/apps on github for updates

> Hello Today we depend to @TheGammaSqueeze for the upgrade of various app like in GammaOS Lite
> 
>     * retroarch 1.15.0 on GammaOS but 1.16.0 is available
> 
>     * Daijishou 1.4.62 but 1.4.67 is available
> 
>     * magisk (but not necessary)
> 
> 
> I thing this is only an issue with the lite version GammaOS with google service, app upgrade may be automatic ? if someone can confirm
> 
> So I thinking about for the next build install f-droid and install retroarch from here It's not possible for now with daijishou [TapiocaFox/Daijishou#513](https://github.com/TapiocaFox/Daijishou/issues/513) (may be just install the apk ?)
> 
> The main purpose is to avoid #64 for exemple and keep retroarch up to date

you can use aurora store (same apps as play store but make sure to login with google acc) or obtanium for emulators/apps on github for updates

poc-sm, 2024-06-11:
Hello
Aurora store is fucked up 
https://gitlab.com/AuroraOSS/AuroraStore/-/issues/1073
So I don't use / recommended  it
I just only use f-droid and I don't have google account 
Obtainium seems a good choice ;-)

poc-sm, 2025-01-24:
Hello @TheGammaSqueeze 
Any updates about this issue ? 
When we start a game in retroarch we have a warning about outdated emulator (when retroachievement is activated) support of retroachievement will disappear in april if we don't upgrade (but we will lose the custom config if we updated on our own) 
ex: https://www.reddit.com/r/RetroAchievements/comments/1hwe5a3/error_outdated_emulator/ 
Obtainium seems to be our best choice for now