TheGammaSqueeze/GammaOS #113: Can't flash on RG405V

ioriyagami79, 2024-01-05:
Hello everyone. 
I did all the previous steps and now I'm stuck in the final step of flashing with gamma os.Opening the .bat file in the folder that I downloaded, seems everything is going on but the process seems stuck at this point (copy/paste what I see in the prompt window):

In attesa da  0 secondi. Per uscire, premere CTRL+C...
fastboot devices
fastboot flash vbmeta_a flash/vbmeta_custom.img

Norzman, 2024-01-09:
I was having similar issues updating to 1.5.1 but realized I was doing `adb reboot bootloader` instead of `adb reboot fastboot`

RoqueAlonso, 2024-01-13:
I have the same issue, but on a RG405M

```txt
13/01/2024 
18:30:43,12 
  
< waiting for device >
^C13/01/2024 
18:42:30,63 
  
< waiting for device >
^C13/01/2024 
18:52:47,49 
  
< waiting for device >
^C< waiting for device >
^C13/01/2024 
18:57:12,12 
  
< waiting for device >
^C< waiting for device >
^C< waiting for device >
^C13/01/2024 
19:10:02,19 
  
< waiting for device >
^C13/01/2024 
19:32:14,19 
```

I`ve waited some time but it seems it is stuck at "fastboot flash vbmeta_a flash/vbmeta_custom.img", not recognicing the device (the device is on the fastbootd screen)

¿Any solutions?

RoqueAlonso, 2024-01-13:
Nevermind, in my case I assumed there were no other fastbot.exe in my PC, but I had.

ioriyagami79, 2024-01-14:
Still have the same issue, can't flash gammaOS

closed 2024-02-23 by TheGammaSqueeze