TheGammaSqueeze/GammaOS #115: Random crashes...

informadortech, 2024-01-08:
Since I installed GammaOS my RG405M crashes randomly, but frequently. There is no behaviour that I can replicate, but I don't even have to do anything, just opening any app, browse on settings menu, it crashes.

I thought this might be fixed with latest 1.5 release but the issue persists.

I read somehwere on reddit that this might be caused because of sd card reader driver, but haven't found a solution.

Already tried with different brand cards, I even tried to go bet to original firmware, but even on stock OS the issue persists.

And this didn't happened before I flash gammaOS.... any solution? I have my RG405M sitting useless on a drawer because of this....

dfreivald, 2024-01-19:
It sounds like a defective board. As a last resort you should try the unbricker tool to return to stock, and then test there. I had similar stability issues, and that worked for me.

informadortech, 2024-01-23:
> It sounds like a defective board. As a last resort you should try the unbricker tool to return to stock, and then test there. I had similar stability issues, and that worked for me.

I don't believe so, the console was working flawlessy before I flashed GammaOS. I have read on reddit something related with the sd card reader that gamma uses, but cannot find that post anymore.

Nevertheress, it keeps happening even if I go back to stock OS. 

I suspect that also can be something related with the EraseUserData script because I always get error when running it. Can't really understand why: I do exact the same as stated on the tutorial...

dfreivald, 2024-02-15:
Errors are normal when running EraseUserData - that's why the first message is "Ignore any errors."

TheGammaSqueeze, 2024-02-23:
Closing due to inactivity. Please let me know if the issues still persist. 
If it does, please enclose a adb logcat log.

closed 2024-02-23 by TheGammaSqueeze

informadortech, 2024-02-23:
> Closing due to inactivity. Please let me know if the issues still persist. If it does, please enclose a adb logcat log.

Yes, it does. I have a lot of random crashes. I already tried to flash original firmware using the unbricker tool to fix partition layout, and also reflash gammaOS but the issue persists. How can I help you to debug?