TheGammaSqueeze/GammaOS #119: Daijishou crash after update to 1.5.1

montero3006, 2024-01-12:
Hello, first of all thanks for that awesome software that is GAMMAOS. I'm using it for months and all is perfect. I updated to 1.5.1 and now have daijishou restarts on RG505 when I select a game with a long press on touchscreen to see the info of the game or to add to favourites.
Here is a video of the problem I uploaded, someone says that have same problem on Odin2 , other says that on RG405M not have the problem.

https://www.reddit.com/r/ANBERNIC/s/VbLWqVmiKM

Thanks for all!

poc-sm, 2024-01-15:
Hello
I can't reproduce the issue 
May be a clean reflash will help ?

montero3006, 2024-01-15:
Hello, thanks for the answer and support. I was looking more on the problem and is not a gamma OS problem I think, on daijisho GitHub lot of people had the problem too. I think is related to some android or Google update because lot of people repair it cleaning the Google play service cache and restarting the machine, normally on Odin2 console.

Here is the issue I opened on daijisho github
https://github.com/TapiocaFox/Daijishou/issues/606

But there people answer me and close the  issue at the moment.

The problem is that for me the bug still there, I done all solutions I see on that forum, clean data and cache from Google play service, from play store, from Google play store and removing, installing them etc etc..

Im sure that your solution for a clean install sure repair the problem, because a clean install sure works because all new and clean. I will finish some started games , make some backups and will do a clean installation.

Thanks again for that awesome firmware, it makes my console works fast nice and looks really cool!

TheGammaSqueeze, 2024-02-23:
Closing due to inactivity.

closed 2024-02-23 by TheGammaSqueeze