TheGammaSqueeze/GammaOS #121: Rumble motor not has drivers in duckstation and aethersx2

Ninoh-FOX, 2024-01-13:
I can't set rumble motor in dualshock console mapped.

dfreivald, 2024-01-19:
Yes, this isn't a GammaOS issue. Assuming the app supports rumble (AetherSX2 works) - You need to set rumble to the stock native Android vibration motor.

Because Gamma implemented a controller driver, it may be possible to forward the rumble API calls to the stock kernel driver in the future, but I would think that's a low priority, given that rumble works already.

TheGammaSqueeze, 2024-02-23:
In the touchscreen settings for Aether and Duckstation, there should be a category to enable vibrations. Use this option, and it'll redirect vibration to the Android OS vibration API.

closed 2024-02-23 by TheGammaSqueeze