TheGammaSqueeze/GammaOS #129: Install GammaOS on microsd card

RingK27, 2024-01-18:
As the title says, is this possible? Will it be? I looked around on reddit/google and can't find info about this topic.

dfreivald, 2024-01-19:
So recent versions of android seem to like a lot of disk I/O. Performance would be severely compromised if you try running it off an SD vs EMMC. Technically it's probably not even possible, without Kernel source code. GammaOS uses the stock kernel from Anbernic.
Why would you want GammaOS running off the SD?

dfreivald, 2024-01-19:
Oh! you're talking about the RK3566 devices (353M/353P/353V). BSP Kernel source is available for those devices, and work is being done to get mainline Linux kernel working. So it is within the realm of possibility to also get a mainline android kernel working, with a ton of effort. So possible yes, probable? I don't know.

RingK27, 2024-01-19:
ok, thank you!

TheGammaSqueeze, 2024-02-23:
The performance will be too poor for the Android system to run from an SD card for the RK3566 series. It's already struggling as it is. No plans to move this to the sd card for now.

closed 2024-02-23 by TheGammaSqueeze