TheGammaSqueeze/GammaOS #130: Bluetooth audio latency

patrislav1, 2024-01-22:
After upgrade from GammaOS v1.3.3 to v1.5.1 the audio latency with Bluetooth headphones is so bad that it's almost unusable (like, half a second). I can reproduce it with AirPods Pro as well as a Bose NC700.

I can't find the exotic 2.5" cable for the Bose so I can't test in cabled mode, but with the internal loudspeaker the latency is fine. So maybe with the audio latency improvements for GammaOS 1.5 the bluetooth audio went south?

Is there an easy way to increase the overall audio latency (like, back to GammaOS 1.3 levels) in case the bluetooth connection needs a bigger buffer to work properly?

patrislav1, 2024-01-22:
> Is there an easy way to increase the overall audio latency (like, back to GammaOS 1.3 levels) in case the bluetooth connection needs a bigger buffer to work properly?

Ok, seems that increasing the audio buffer from 96 to 480 samples does the trick.

```diff
--- audio_pcm.xml.orig  2024-01-22 17:35:01.224072606 +0100
+++ audio_pcm.xml       2024-01-22 18:24:49.905174254 +0100
@@ -7,7 +7,7 @@
                <voip channels="1" rate="16000" period_size="640" period_count="2" format="0" start_threshold="1280" card="sprdphone-sc2730"  device="6" FE_device="FE_ST_VOIP"/>
                <fm channels="2" rate="32000" period_size="1280" period_count="2" format="0" stop_threshold="-1" card="sprdphone-sc2730"  device="7" FE_device="FE_ST_FM"/>
                <fm_bydsp channels="2" rate="32000" period_size="1280" period_count="2" format="0" stop_threshold="-1" card="sprdphone-sc2730"  device="15" FE_device="FE_ST_FM_DSP"/>
-               <fast channels="2" rate="48000" period_size="96" period_count="4" format="0" card="sprdphone-sc2730"  device="0" FE_device="FE_ST_NORMAL_AP01"/>
+               <fast channels="2" rate="48000" period_size="480" period_count="4" format="0" card="sprdphone-sc2730"  device="0" FE_device="FE_ST_NORMAL_AP01"/>
                <dsp_loop channels="1" rate="8000" period_size="640" period_count="4" format="0" card="sprdphone-sc2730"  device="10" FE_device="FE_ST_LOOP"/>
                <voice_tx channels="1" rate="16000" period_size="640" period_count="2" format="0"  start_threshold="1280" card="sprdphone-sc2730"  device="53" FE_device="FE_ST_VOICE_PCM_P"/>
        </playback>
```

I used the method from here: https://www.reddit.com/r/retroid/comments/12jdp41/fixed_the_latency_issue_with_the_rp3/

Would it be possible to make a UI element to let users fine-tune the audio buffer size for their particular use case, from a settings menu?

TheGammaSqueeze, 2024-02-23:
The UI element is not possible as this file is read only upon device boot. Interesting that increasing the buffer works for you though. I'll keep this issue open for others for visibility.

Ace20xx, 2024-02-28:
Here is my experience with different BT devices on Gamma OS 1.5.1, no setting modified and tested by earbud delay test from the Play Store.

Samsung Galaxy Bud Pro / Shokz openrun pro
Latency about +400ms

Sony WF-1000XM4
Latency about 0 - +50ms