TheGammaSqueeze/GammaOS #133: Analog stick sensitivity

patrislav1, 2024-01-27:
With GammaOS v1.5.1 the selectable analog stick sensitivity is "normal", -15%, -25%, -50% and "custom".

I'm at a point in a game where the native sensitivity is much too high, -15% and -25% is still too high but at -50% I can't move fast enough with the stick so would need something in between.

Is there a way to modify the pre-set values for the sensitivity menu, or to set a value for the "custom" setting?

dfreivald, 2024-02-02:
https://github.com/TheGammaSqueeze/retrogame_joypad_2_xbox/blob/83b2b5807d9a925cad1cf892b9f4ca337f74af68/retrogamepad2xbox.c#L1226

patrislav1, 2024-02-03:
> https://github.com/TheGammaSqueeze/retrogame_joypad_2_xbox/blob/83b2b5807d9a925cad1cf892b9f4ca337f74af68/retrogamepad2xbox.c#L1226

Thank you, will try that.

closed 2024-02-03 by patrislav1