TheGammaSqueeze/GammaOS #134: Bricked 405M after following instructions can anyone assist with recovery?

Strappingyoungstu, 2024-01-27:
Hi so I've followed the instructions and got to the point of flashing, that went OK then went to Erase User Data and despite saying it was going to take 10 minutes it only took a few seconds as it said it couldn't find a particular file... nevertheless I ignored the error messages and now the device wont boot. 

I have downloaded the 7z files from the Black Seraph page but after extracting I just get 7z files... not sure what to do from here or where the readme file is supposed to be in 001

Any advice appreciated!

HenryAvery, 2024-01-28:
I have a video on how to unbrick the device if you want to try to follow that. 
https://youtu.be/XKKzShK7JAY

TheGammaSqueeze, 2024-02-23:
Closing due to inactivity

closed 2024-02-23 by TheGammaSqueeze