TheGammaSqueeze/GammaOS #135: SyncThing Integration

Bent3ndo, 2024-02-04:
Hopefully it is possible in future updates to include a cloud syncing option such as SyncThing or a Google Drive plugin (similar to EmuDeck) to allow multiple devices to share cloud saves across multiple OS'.

poc-sm, 2024-02-05:
Hello
It's delegated to end user 
If you need you can install sync app as you want from play store / f droid / aurora / apk 
We have a lot of server app so I think it's not relevant to include them all

Bent3ndo, 2024-02-05:
What is the system EmuDeck puts in place that allows you to choose DropBox, Google Drive, OneDrive etc and have a command line automatically run a routine save folder upload? I'd have thought that's something GammaOS would be capable of and therefore give multiple devices such as Anbernic handhelds the ability to cross-save automatically.

poc-sm, 2024-02-05:
I don't know emudeck but already play with Jelos and we have syncthing and Rclone I thinks it's something similar
If I check the doc, yes it's use syncthing 
https://emudeck.github.io/community-creations/steamos/tools-and-guides/#how-to-set-up-syncthing 
So all you need is to download syncthing app and configure it

TheGammaSqueeze, 2024-02-23:
Closing due to inactivity.

closed 2024-02-23 by TheGammaSqueeze