TheGammaSqueeze/GammaOS #138: Google Play doesn't work

Bent3ndo, 2024-02-09:
Hi; I've tried using the Google Play store multiple times, and have tried removing all updates etc as well, and no matter what I do the Play store refuses to context.

dfreivald, 2024-02-15:
Any more info? 

- Which version of GammaOS
- Is this a fresh install
- Did you run wipeuserdata

Bent3ndo, 2024-02-18:
The latest version of GammaOS for the RG405V (has been the case ever since I moved from stock firmware however so hasn't just been the current version).

The beta of GammaOS was a clean install and then I updated through the official way.

And I've not run any erase options or programs at any point since GammaOS has been installed. The most I've done it remove updates to see if reverting back to the original version included helped, but it doesn't.

dfreivald, 2024-02-20:
I would use the unbricker to restore the stock OS, and then re-flash GammaOS. Then you have to run eraseuserdata, it's a required step if you want the play store to work.

TheGammaSqueeze, 2024-02-23:
Consider disabling Private DNS if you are having issues with connectivity.

https://github.com/TheGammaSqueeze/GammaOS/issues/142#issuecomment-1961179789

closed 2024-04-04 by TheGammaSqueeze