TheGammaSqueeze/GammaOS #139: 405v  crashes and shuts off at 50-65% battery

Ogsmoketho92, 2024-02-11:
Gamma issues

I flashed my 405v weeks ago and I noticed that While paying any game with  high performance switched on (sometimes even on normal mode) it would just shut off on me .. it usually happens when it reaches about 50-something percent  it’ll crash and shut off!  After booting it back up and make it pass the Gamma logo it shuts off and it continues to do the same until I throw it on the charger..now I’m not 100% sure if it’s the OS or if I’m dealing with a faulty battery 🥵🫠 but then again I don't remember ever happening on the first release of gamma for the 405v only after upgrading to 1.5.1

ANYONE ELSE EXPERIENCING THIS ISSUE?

lucasgomeshora222, 2024-02-13:
Same here on my Rg505, I bought a replacement battery from AliExpress, waiting to test...

lucasgomeshora222, 2024-02-18:
> I think we need a FAQ, because this would be question #1. Battery calibration is in your user data, so if you wipe your user data, do a factory reset, unlock or lock the boot loader, it requires a full drain and charge cycle with the OS booted, to recalibrate the battery data. I've heard it may take a couple cycles, but I've only ever needed one on my devices.

But the question is: how can I drain the entire battery if the device turns off when the battery is at 50-60% percentage?

MissingNoodles, 2024-02-19:
Question; Are you all on 1.5 or 1.5.1? 

I've also had a similar issue where my RG505 randomly shuts down at 50% or drains the battery quick after 25%. I've been on 1.5 for over a month now, did the battery calibration twice and everything worked ok (minus hitting 25%) until late last night when I got to about 50% and my system shutdown while playing ps2 on the normal performance setting. The system would not power back on until I connected to power adapter while pressing power on. When I got back in the system, my battery showed about 49% (if I remember correctly). Currently testing to see if the system will shutdown again at 50%

Will a bug report help anyone?

dfreivald, 2024-02-20:
> But the question is: how can I drain the entire battery if the device turns off when the battery is at 50-60% percentage?

If the device is shutting off, it probably is drained. Recalibrating is what gets the percentage to show correctly.

lucasgomeshora222, 2024-02-20:
> Question; Are you all on 1.5 or 1.5.1?
>
> I've also had a similar issue where my RG505 randomly shuts down at 50% or drains the battery quick after 25%. I've been on 1.5 for over a month now, did the [...]

I'm on 1.5. I can't remember if this happened on the 1.3...

dfreivald, 2024-02-21:
It should have happened to most as soon as they unlocked their bootloader, or wiped their user data.

MissingNoodles, 2024-02-23:
> I'm on 1.5. I can't remember if this happened on the 1.3...

I got my second shutdown at 60% again today, but yesterday I was able to through it just fine. I am going to update to 1.5.1 to see if that helps.

TheGammaSqueeze, 2024-02-23:
When the device shuts off, try unplugging the battery, plug it back in.

Then turn the device back on with a usb cable to your pc plugged in. Then let it charge overnight (making sure Android is still booted). The calibration when Android is running vs the console is turned off when charging is different for some reason.

MissingNoodles, 2024-02-23:
> When the device shuts off, try unplugging the battery, plug it back in.
>
> Then turn the device back on with a usb cable to your pc plugged in. Then let it charge overnight (making sure Android is [...]

I will give this a try. Thanks.

lucasgomeshora222, 2024-02-25:
> Is there any chance it's a hardware issue? I've tried everything you mentioned but nothing works ={

I tried to solve the problem by doing a factory reset but now I'm stuck on a screen that says Wiping data --- Formatting data. Can anyone help me revert it?
[image]

MissingNoodles, 2024-02-26:
> > When the device shuts off, try unplugging the battery, plug it back in.
> > Then turn the device back on with a usb cable to your pc plugged in. Then let it charge overnight (making sure Android is still booted). The calibration when Android is running vs the console is turned off when charging is different for some reason.
> 
> Welp- My device shutoff again at around 50% today and I opened her up, removed the battery for a minute, reconnected the battery, connected the power USB to my PC, and powered the RG505 on. This time I got corrupted system error(see attached). Should I re-image 1.5 or go up to 1.5.1?
> 
> Excuse my McDonalds overlay. It was a joke that went too far.
> 
> Thanks again for your help.
> 

Not sure if this will help anyone, but after I got stuck on the android recovery with the corrupt data message, I let my device charge for a bit, then I connected it to my PC, opened CMD and did `adb reboot fastboot` to get me into fastboot on the RG505. From here, I ran the 1.5.1 partition script, waited, then ran the erase user data script, waited, and then rebooted back to normal. I lost all my data of course.

I did try adb pull and a couple of adb file explorers to see if I could recover any data, but no good. Bummer!

closed 2024-04-04 by TheGammaSqueeze