TheGammaSqueeze/GammaOS #142: Limited WiFi Connectivity after flashing Gamma OS (either full or lite version)

sAmenta, 2024-02-22:
With StockOs, I didn't experience any of the issues I'm about to describe. 
I have a RG405v, and I used the 405v firmware for flashing Gamma.
When I flashed the full version of GammaOs, I was stuck on the Google setup wizard because the WiFi connection appears to be limited and I can't connect to the internet. I'm based in China, and I used to have this same issue with my Quest 2 VR headset, but I was able to overcome it by sharing my VPN across the network.
However, this workaround doesn't work with GammaOs full version and I can't get rid of the limited connectivity issue.

Then I tried GammaOs Lite and I was able to move on, but my WiFi network still shows limited connectivity. In this case, sharing the VPN across my home network helped me connect to the internet. It's not very ideal because I wouldn't be able to connect to the internet from my RG device whenever I'm outside of my home network

Other than that, I also experience problems launching apps in Gamma Lite. For example, when I click on the Dolphin Emulator app, it launches and then crashes right away. And sometimes the Aurora store shows me an "error parsing the package" when I try to install an app...

TheGammaSqueeze, 2024-02-22:
Have you tried disabling private DNS?

sAmenta, 2024-02-22:
Hi, thanks for the tip. I can have a try on that.

How about the parse error with Aurora store and some apps failing to open? For example, I tried the dolphine emu that comes with Gamma and when I click on it, it shuts down right away. I got the same issue with other apps I installed from Aurora, such as the Airdroid app.

TheGammaSqueeze, 2024-02-22:
Dolphin emulator does not come with GammaOS. You need to install this
manually.

Are you sure you ran the Eraseuserdata script before you booted into
GammaOS?

sAmenta, 2024-02-22:
yes, I followed your guidelines step by step and also checked the videos. In both cases, when I flashed GammaOS and GammaOS Lite, I also run the Eraseuserdata.bat

TheGammaSqueeze, 2024-02-22:
You may also try the factory reset option within Android itself.

With that being said, I am unable to reproduce the issue.

sAmenta, 2024-02-22:
I will try the factory reset.

For the GammaOS full, is there way to disable the private dns during the Google Setup Wizard, or is there a way to skip the wizard  and do it later after I successfully enter the OS?

TheGammaSqueeze, 2024-02-23:
You can use ADB on your computer to do this. Connect your 405 via USB, open a command prompt window, issue the following commands:

adb shell 
settings put global private_dns_mode "off"

closed 2024-04-04 by TheGammaSqueeze

lean-ace, 2024-05-15:
[wifi.log](https://github.com/TheGammaSqueeze/GammaOS/files/15323688/wifi.log)

I have the same problem, i uploaded the log.

DeathyYoung, 2025-07-14:
> You can use ADB on your computer to do this. Connect your 405 via USB, open a command prompt window, issue the following commands:
> 
> adb shell settings put global private_dns_mode "off"

it works for me, my device is a 405M