TheGammaSqueeze/GammaOS #143: Back Button Customisation

Bent3ndo, 2024-02-23:
Now that ES-DE is available as a front end for Android, when attempting to back out of a game using it as the front end, rather than backing out to ES-DE it backs into the app list.

Could the use of the back button be altered as a hot key in a way that enables it to back out into the previous app? That way RetroArch/Dolphin/etc would back straight into ES-DE again instead of the OS menu.

Thanks.

TheGammaSqueeze, 2024-02-23:
This is not a GammaOS issue. This isn't possible as ES-DE is not a launcher app.

You should follow RetroGameCorps guide on ES-DE setup on how to do this with workarounds and multiple third party apps.

For clarity, the normal behaviour is for the emulator that is launched to go back to the previously opened app if you exit the emulator from within the app itself.

Holding the back/home button for 3 seconds acts as pressing the home button taking you back to the default launcher.

In RetroArch I've already configured this so that the app closes instead of taking you back home. However holding the back button in any other app will cause you to go back to the launcher.

https://github.com/TheGammaSqueeze/GammaOS/assets/116582950/0a8afd2e-f533-4b45-9e09-85fba7182b10

Lamcza, 2024-03-04:
> This is not a GammaOS issue. This isn't possible as ES-DE is not a launcher app.
> 
> You should follow RetroGameCorps guide on ES-DE setup on how to do this with workarounds and multiple third party apps.
> 
> For clarity, the normal behaviour is for the emulator that is launched to go back to the previously opened app if you exit the emulator from within the app itself.
> 
> Holding the back/home button for 3 seconds acts as pressing the home button taking you back to the default launcher.
> 
> In RetroArch I've already configured this so that the app closes instead of taking you back home. However holding the back button in any other app will cause you to go back to the launcher.
> 
>  20240221_230536_1.mp4

Can i somehow speed up this 3 sec  or edit the behavior of f/home/back button somewhere in /system/usr/keylayout/  ?

TheGammaSqueeze, 2024-04-04:
> Can i somehow speed up this 3 sec or edit the behavior of f/home/back button somewhere in /system/usr/keylayout/ ?

Not from within GammaOS itself.
You will need to fork the rgp2xbox repo, make your changes, create a magisk module including your new binary and install that.

https://github.com/TheGammaSqueeze/retrogame_joypad_2_xbox/blob/main/retrogamepad2xbox.c

closed 2024-04-04 by TheGammaSqueeze