TheGammaSqueeze/GammaOS #144: I cannot change the default launcher

rcerrejon, 2024-02-29:
Hi, yesterday I installed ES-DE on the RG450m, so far so good, and it is 100% compatible with GammaOS, I just have a small problem, I changed the default Android launcher to "AnyHome" (as Russ says in the RetroGameCorps video), but when I reboot the system, it reverts to Daijisho.
How can I change it permanently?

Thanks.

TheGammaSqueeze, 2024-02-29:
You can use adb on your computer, connected by usb, to disable Daijisho. Then set your other launcher as the default.

Use the following command in a command prompt window (make sure you have adb tools installed as per the GammaOS install guide)

adb shell pm uninstall -k --user 0 com.magneticchen.daijishou

rcerrejon, 2024-02-29:
Thanks, at the moment I don't want to completely uninstall Daijisho, but good idea to use adb, I disabled it with this command:

`adb shell pm disable-user --user 0 com.magneticchen.daijishou`

MaxDjently, 2024-03-26:
I don't find either of these two answers really fix the problem.  I've tried the preinstalled version, the Playstore version, and sideloaded the apk.  There is no difference, Daijisho stays the default launcher as long as it's installed.  Nova Launcher has a nice feature where I can set the Game Launcher button on my RG405V to launch Daijisho and switch between launchers by pushing the button.  But only if Nova Launcher is default.

MaxDjently, 2024-03-26:
Here is what I did to work around it so I can use both Nova and Daijisho.  I downloaded Macrodroid and watched this tutorial to learn how to use it:  

https://www.youtube.com/watch?v=BalggiRDhhA&t=302s&pp=ygUYbWFjcm9kcm9pZCBob21lIGxhdW5jaGVy

But instead of launching Chrome, I clicked "Shell Script," and then entered this into the field:

cmd package set-home-activity com.teslacoilsw.launcher

Then in Nova Settings I went to Gestures & Inputs>Home Button, then the Apps tab and selected Daijisho.

Restarted the phone, and now it boots into Daijisho by default, but the Game Launcher button now switches between Daijisho and Nova.

TheGammaSqueeze, 2024-09-01:
[Gamma_Home_Override.zip](https://github.com/user-attachments/files/16829901/Gamma_Home_Override.zip)
Apply this magisk module to stop the override