TheGammaSqueeze/GammaOS #150: LD39 cable on RG405V (1.5.1 lite)

LHRSilva, 2024-03-15:
Hello, I saw an old video of the Gamma OS with the LD39 cable on the @TheGammaSqueeze channel on YouTube, does the system still support it? Here, I'm using the RG405V with Gamma 1.5.1 Lite and couldn't get it to work, is there any specific configuration needed? sorry... for english by IA =/

LaVecchia1969, 2024-03-30:
Hi friend. I tried to connect my anbernic rg505 to and hdmi output, it works! But..not ever. Explain: i bought from Aliexpress  an usbc to hdmi cable (is the ld39 also it is not written in package, like china traditions). In rg505 gamma os i tried to optimize the battery settings to free use of battery by the app, i tried to consent mirroring app (the app to download for the mirroring) to execute in background e i tried to allow notification from mirroring app, i tried to allows usb debugging and the use of default usb to permit trasmission of files. The result: after a while my rg505  connect the monitor (hdmi output) showing the classic picture of mirroring app. Then one time yes, one time no , i got the message of allowing mirroring app to use usb port as notification of the gamma os in rg505. When it happens, the ld39 cable send the video: all is ok! The problem is that this issue some time is avalaible some time is impossible to obtain. So, the hardware surely support tv out from usb c to hdmi via ld39 , but the software need configuration or tuning (gamma os in new release?) not easy to find. I searched for you tube and internet i saw tons of video, one in russian also, but i never got a way to obtain the result in sure and determined way. For this i not have the possibility of a sure connection to tv. Any contribution is appreciated.

LHRSilva, 2024-03-31:
> Hi friend. I tried to connect my anbernic rg505 to and hdmi output, it works! But..not ever. Explain: i bought from Aliexpress an usbc to hdmi cable (is the ld39 also it is not written in package, [...]

Hello, thank you for sharing your knowledge. I will try to do as you suggested. It's good to know that there are more people interested in this type of cable because with the delay of wireless transmission, it becomes unplayable. I saw that in the original system, the cable works very well. Let's hope for a nod from the developers. Thank you very much.

LaVecchia1969, 2024-03-31:
Hi friend. Are you sure in original system the cable works very well? I saw that also in the original system is not easy to have hdmi out. Often the videos are cutted in the moment of the message like ... allow usb out in mirroring app. Not easy, seems, in the original system also. Tons of user try in but in portuguese language... not easy to understand. Yes they tried to connect with the original system and gamma os have not the same settings screens. Anyway, if  you have news or you find the way let us know. This issue in very important for playin in tv screen. Maybe can be a reason for backing up to the original anbernic android? I hope not! Greeting.

LHRSilva, 2024-03-31:
> Hi friend. Are you sure in original system the cable works very well? I saw that also in the original system is not easy to have hdmi out. Often the videos are cutted in the moment of the message [...]

So I saw on YouTube in the video "Como Ligar o Anbernic RG505 ou RG405M e
Jogar na TV ou Monitor" on the channel "Ruivo ™" in Portuguese, it was the rg405m model in the original system and it seemed to work very well.

LaVecchia1969, 2024-03-31:
i saw this video in portughese, i think the video was cutted because not easy to set the tv out like the video seems show.. now i tried to go back in original anbernic. the cable work better, but the problem remain. no possibilities to get out the signal.  very difficult find a way.  both sistem gamma os and original but no good !

TheGammaSqueeze, 2024-04-04:
I will need to find my cable to test this again, but I am not sure why this is not working for you.

This may be several weeks before I can test this

systemmails3, 2024-04-18:
news. i saw that the anbernic rg556 with a cable sold on amazon works. but for this console is there no news?

nitehack, 2024-10-22:
I'm interested too, but I'm reading that there are a lot of problems. I don't know whether to buy the cable or not. WiFi streaming has a lot of lag