TheGammaSqueeze/GammaOS #154: Joystick Calibration.

richoski7, 2024-03-30:
RG405M - v1.51 - Left joystick only reaches about 80% when pushing up. 100% for other directions and right stick.

TheGammaSqueeze, 2024-04-04:
Try flashing the DTBO from GammaOS 1.3 for your device using fastboot.

dfreivald, 2024-04-08:
Someone on discord got a batch of 405M with out of spec hall sticks with s similar issue on the left stick.
They used GammaOS custom analog mapping txt file to recalibrate/scale up the analog readings.

ITh1erry, 2024-06-19:
> Try flashing the DTBO from GammaOS 1.3 for your device using fastboot.

Which DTBO do you recommend flashing in case of an RG405V on V1.5.1? I just tried flashing the dtbo_custom.img from V1.5.1 again and got the following error:
[image]

ITh1erry, 2024-06-19:
> Try flashing the DTBO from GammaOS 1.3 for your device using fastboot.

To resolve this issue on an RG405V GammaOS 1.5.1, which DTBO should I flash?

ren7o, 2024-06-26:
Hello any news about this problem @TheGammaSqueeze ? I have exactly the same problem on GammaOS 1.5.1 after few days the problem with left stick appeared from nowhere. I tried to reflash original rom and there it works as should.

ITh1erry, 2024-06-26:
@ren7o thanks for verifying this. I was planning to use this as a last resort. It’s good to know that reflashing will work. Have you tried a factory reset of GammaOS also by any chance? I’d still like to stay on Gamma though..

ren7o, 2024-06-26:
No problem @ITh1erry I will update here if I find anything. I tried almost everything like: reset/wipe GammaOS in recovery, reset/wipe in GammaOS, flashing GammaOS 1.3 DTBO (it will break fan, fan spins even if device is in sleep, helped a little but still not usable). I tried reflashing GammaOS also reflashing stock and than GammaOS (problem will pop up again). Nothing works. So I am back on stock now... I love GammaOS but I cannot stay on it because joystick is crucial to me.

ren7o, 2024-06-26:
I have to add I have RG405V but it seems it does not matter.

ITh1erry, 2024-06-26:
@ren7o I also have a RG405V, posted a similar issue like this one in #185 . So I guess it’s back to stock OS for me too. Did you install a custom frontend to be able to install other emulators?

ren7o, 2024-06-26:
@ITh1erry Good, I hope @TheGammaSqueeze will look on this issue. Yes I installed Daijisho and uninstalled both official launchers. I got it rooted so I could debloat it a bit. You can freeze (no root needed) these launchers as well but peoples reporting they unfreeze randomly or after restart sometimes. You have to remap the home button as well with some app remapper (look for F10 button).

ToCarlosWebos, 2024-07-29:
I'm having exactly the same issue, it's absolutely not a hardware related problem, if you just go into stock OS the left Joystick will work again just fine, gammaOS is still very faulty for the rg405v, it's not worth for now, maybe in a year

dexwerx, 2024-07-29:
The stock OS works well for bad sticks, by doubling the scaling. This way you reach full magnitude by only pressing the stick half way to the edges, and heavy cardinal snapping.

ren7o, 2024-08-26:
@dexwerx hey, this is not hardware related issue. We have not bad sticks, it seems the hardware has been silently updated or something... The same like with the wifi problem.

dexwerx, 2024-08-28:
@ren7o - Luckily Gamma gave us the option to use custom scaling. You can use GammaOS and map the calibration to the same scaling as stock. Let me know if you need help creating the CSV for your "updated" hardware.

dexwerx, 2024-08-28:
Here's some PowerShell to create a custom scaling closer to stock.
```
$csv   = 'DecreaseAnalogSensitivityCustom.csv'
$max   = 1800
$scale = 0.80
$map   = [Collections.ArrayList]::new()
foreach ($i in -$max..$max) {
    $o = [Math]::Min([Math]::Max([Math]::Round($i / $scale), -$max), $max) 
    [void]$map.Add("$i,$o`n")
}
sc $csv ($map -join '') -NoNewline
```

use a root file explorer to copy the resulting CSV to /data/rgp2xbox/
tap the quick toggle and set it to your custom calibration

Please don't forget to update us with your results.

[DecreaseAnalogSensitivityCustom.csv](https://github.com/user-attachments/files/16786700/DecreaseAnalogSensitivityCustom.csv)

dexwerx, 2024-08-28:
And for convenience here's a CSV with 70% scaling.

[DecreaseAnalogSensitivityCustom.csv](https://github.com/user-attachments/files/16786884/DecreaseAnalogSensitivityCustom.csv)

ren7o, 2024-08-28:
@dexwerx wou, thank you! I will look in to this later because I am freshly on stock OS... But there are others who can test it for us.

dexwerx, 2024-08-29:
Trying to work around the stick variations is not easy, because driver is mostly closed source, and the scaling is embedded in the device tree. GammaOS is using what was established in other CFW (ArkOS/JELOS/RockNIX) to be known good scaling, but unfortunately it's causing issues with recent devices. Using scaling values that are too small, crushes the range sensitivity for everyone, but hides out of spec analogs. This is why I believe Anbernic will never fix the analog calibration on Stock OS.

In order to maximize the analog range for everyone, the best we can do currently is to keep the current scaling values, and use recalibration for the analogs that fall short on range. 

Ideally in the future we could use a GUI to adjust scaling and recalibrate the sticks independently, as well as use non linear curves. Gamma made the source for rpg2xbox available so anyone with the skills could help build an app to facilitate this.

Nukeshadow, 2025-08-06:
It's been... nearly a year?  Going to look to see if there's a more recent thread, but yeah... having the same issue with GammaOS Next installed on a Cube.  Going to do my best to get it working without re-flashing stock but... not going to tolerate a janky stick.  It's not like I'm using this thing to do my taxes.  If a gaming system can't game, the plot has been lost.