TheGammaSqueeze/GammaOS #155: RG505 No wifi after Gamma install

bcrosby1978, 2024-04-01:
I've tried this like 4 times now, using the 505 full, and lite files.  Install goes great, no errors.  When I get to the setup screen it just spins on finding a network. I dont get what is wrong.

dtimber, 2024-04-03:
Same issue here. Tried the lite version first and then the full version but still couldn't find a WiFi network.

Classic-Jack, 2024-04-03:
Same issue here, just got the device today, flashed the CFW and now no WiFi. Can do most of what I want on it without it but it has WiFi so I want WiFi. Hope all is well over on the Devs side if this is a dev issue

mrkushaljoshi, 2024-04-04:
Logged Issue as well: [#157](https://github.com/TheGammaSqueeze/GammaOS/issues/157)

TheGammaSqueeze, 2024-04-04:
Hi all.

Nothing has changed in the GammaOS firmware that would cause this. It is out of my control if my assumptions below are correct.

This may be something to do with the Anbernic base firmware potentially being different compared to when GammaOS was released.

Typically, you should be able to resolve this by using the 505 unbricker, then reflashing GammaOS.

Please confirmed if you have tried this.

If this doesn't resolve your issues, then this may be a hardware revision but I doubt this.

bcrosby1978, 2024-04-04:
Yes, I was able to go back to stock OS using the unbricker and wifi worked.  I will note after installing Gamma my device name in the system said RG405V, instead of RG505.  If this is an issue due to a new version of the 505, is that something that will be addressed?

dtimber, 2024-04-04:
@TheGammaSqueeze : Thanks for the reply. As you suggested I reflashed stock firmware using 505 unbricker. Wifi then worked again within stock firmware. After this i've flashed GammaOS lite again. Now Wifi also works within GammaOS lite.

It seems that you guessed right that Anbernic did some changes to the base firmware on newer devices that prevent Wifi from working.

Thanks again for your quick response and also for all the effort you put into GammaOS.

bcrosby1978, 2024-04-04:
> @TheGammaSqueeze : Thanks for the reply. As you suggested I reflashed stock firmware using 505 unbricker. Wifi then worked again within stock firmware. After this i've flashed GammaOS lite again. [...]

How did you manage to get gamma lite working?  You said it didn't work at first.  What did you change?  What firmware are you on?

TheGammaSqueeze, 2024-04-04:
> Yes, I was able to go back to stock OS using the unbricker and wifi worked. I will note after installing Gamma my device name in the system said RG405V, instead of RG505. If this is an issue due to a new version of the 505, is that something that will be addressed?

This is just a bug with GammaOS Lite, I forgot to update the name, can safely ignore.

> @TheGammaSqueeze : Thanks for the reply. As you suggested I reflashed stock firmware using 505 unbricker. Wifi then worked again within stock firmware. After this i've flashed GammaOS lite again. [...]

Thanks for confirming!
I'll leave this issue open for visibility.

dtimber, 2024-04-04:
> How did you manage to get gamma lite working? You said it didn't work at first. What did you change? What firmware are you on?

I've flashed stock firmware using 505 unbricker. Verified that WiFi was working on stock firmware. Then flashed GammaOS lite 1.5.1.

mrkushaljoshi, 2024-04-04:
> > > @TheGammaSqueeze : Thanks for the reply. As you suggested I reflashed stock firmware using 505 unbricker. Wifi then worked again within stock firmware. After this i've flashed GammaOS lite again. Now Wifi also works within GammaOS lite.
> > > It seems that you guessed right that Anbernic did some changes to the base firmware on newer devices that prevent Wifi from working.
> > > Thanks again for your quick response and also for all the effort you put into GammaOS.
> > 
> > 
> > How did you manage to get gamma lite working? You said it didn't work at first. What did you change? What firmware are you on?
> 
> I've flashed stock firmware using 505 unbricker. Verified that WiFi was working on stock firmware. Then flashed GammaOS lite 1.5.1 again.

Likewise! I was able to use the unbrick tool to get stock firmware. I think validated WiFi worked fine. I flashed GammaOS lite where WiFi worked but I had some issues with Google Play Services. I then flashed GammaOS (non-lite) and then did a factory reset via settings. Everything looks like its working now, I did just take delivery of my device yesterday so my unit is newer, seems to be a issue plaguing new RG505 devices.

dfreivald, 2024-04-06:
Just in case it wasn't clear - the fix is to 
* use the unbricker to revert to stock and a partition the GammaOS installation expects
* Reflash GammaOS

bcrosby1978, 2024-04-06:
What partition are you talking about?  How do you do that?

bcrosby1978, 2024-04-06:
I did use the unbrick tool....3x...and tried to reinstall gamma...no luck

dtimber, 2024-04-06:
@bcrosby1978 : When you used 505 Unbricker did WiFi work within stock firmware?

bcrosby1978, 2024-04-06:
Yes...every time I've went back to stock....the wifi fired right up.

dtimber, 2024-04-06:
OK and you tried both full and lite version?

bcrosby1978, 2024-04-06:
Yep..tried both 2 times

dfreivald, 2024-04-07:
So far the unbricker worked for AverageGamer, dtimber and mrkushaljoshi, but it's still not working for bcrosby1978.

bcrosby1978, 2024-04-07:
Could it be a firmware issue?  I got mine last week directly from Ambernic...my buddy got his from Amazon.   His worked first time. I updated to 1.24 before my install.

dfreivald, 2024-04-07:
It shouldn't be, GammaOS is using the exact same kernel from the stock android. If Wifi works using the unbricker, it should work in GammaOS.

PanteraProductions, 2024-04-08:
> I've flashed stock firmware using 505 unbricker. Verified that WiFi was working on stock firmware. Then flashed GammaOS lite 1.5.1 again.

Made an account just to confirm, flashing gamma - unbkricker - back to gammaOS works. At least for me.

My RG505 came in a couple of days ago, thank you for the help!

tml89, 2024-04-13:
Hello!
Any progress on this?
I'm on Linux only so not able to unbrick my rg505.

dfreivald, 2024-04-14:
Unfortunately the Spreadtrum/Unisoc updatedownload tool only runs on windows.

TheGammaSqueeze, 2024-05-01:
It would be useful if people could provide logcat logs when it comes to wifi issues.

Maybe use a tool like this to save the logs
https://github.com/JonathanDotCel/LessShittyLogcat

Mr-BrightsideG, 2024-05-02:
Hello, I have had exactly the same problem when flashing GammaOS, in my case I have an rg405v (it was purchased directly from the manufacturer) and it does not allow progress from the first boot, since it does not detect any wifi signal, I was reading your discussion and you mentioned a possible revision of the console, I wouldn't know about the internal components, however I have seen that the latest rg405v come with a protector on the fan to avoid noise due to friction of the fan and that many users complained at the time, which although it is not a very rigorous revision, it gives an indication that something may have changed.

I have tried going back to Stock and trying flashing to GammaOS again, however it did not work, it is worth mentioning that with Stock OS everything works without problems.

It's a shame, since despite not being able to enjoy GammaOS, I could see the improvement on the screen by eliminating the sharpness that StockOS brings, I also really liked the work and time that goes into it, despite the problems, thank You very much to the developer for the effort and I hope a solution can be found. If you have any questions, I am willing to help.

akilaweerat, 2024-05-04:
I can also verify that I used unbricker for rg505 using this guide -> https://youtu.be/XKKzShK7JAY?si=D2tom_zVC9_oNv2g 
 which installed a older firmware of the stock os.

and then I reinstalled gamma os , and it worked :)

Mouer78, 2024-05-07:
I am having the same WiFi issue with the rg405v.
Both versions of Gamma result in no WiFi. 
I did make sure to use Gamma 405v version. 
Must have flashed the firmware, unbricked, reflashed gamma 3 or 4 times before finally giving up.
It's a shame because I love the performance boost that comes with Gamma for PS 2 games!

CdadM, 2024-06-25:
Is there any updates on progress solving this? In case this helped, received my rg405v this week, wifi works on stock but not on gamma. I've been forth and back 4 times by now, tried changing cable, USB port, connecting to wifi when back on stock, rebooting, powering off... Nothing until now. I also tried to eliminate the vendor custom line from the .sh file, nothing. 
Any ideas?

CdadM, 2024-06-25:
Thanks, looking forward to it

GMbizi, 2024-06-30:
> Anbernic keep insisting that nothing has changed from their end.
> 
> _Doubt_
> 
> I've worn them down into sending a new 405v from factory, once I have this, I'll be able to see what the problem is for these new units.
> 
> This will be a couple of weeks

I have the same problem for an rg405v received yesterday, no wifi, I managed to use my phone's internet connection with Bluetooth but the quality is average, I hope the problem can be resolved with an update , which I am impatiently waiting for

I wonder if it's because I never used wifi in stock os and switch quickly to gamma os ?

TheGammaSqueeze, 2024-06-30:
Until we get to the bottom of the issue, I've put a warning on the Github page to this issue for new units.

I wish I could fix this, but I need the new hardware in front of me. I'm waiting on Anbernic.

rikmcastro, 2024-06-30:
Looking forward to an update on this issue.

CdadM, 2024-07-02:
Well, I said 

> Anbernic keep insisting that nothing has changed from their end.
> 
> _Doubt_
> 
> I've worn them down into sending a new 405v from factory, once I have this, I'll be able to see what the problem is for these new units.
> 
> This will be a couple of weeks

So, my case is resolved! I said I edited the .sh file and no luck, but I tried editing the .bat directly and voilà, all worked like a charm. Thanks @TheGammaSqueeze for all your work

rikmcastro, 2024-07-02:
That's awesome! Would there be a way to make a step by step guide available?

Kamins2017, 2024-07-03:
> Well, I said
>
>
> So, my case is resolved! I said I edited the .sh file and no luck, but I tried editing the .bat directly and voilà, all worked like a charm. Thanks @TheGammaSqueeze for all your work

This also worked for me, thank you!

To everyone wondering, delete the following line from both the FlashPartitions SH file and also the FlashPartitions .bat file
"fastboot flash vendor flash/vendor_custom.img >>log.txt 2>&1"
Use NotePad to edit both and make sure you hit save.

rikmcastro, 2024-07-03:
> This also worked for me, thank you!
>
> To everyone wondering, delete the following line from both the FlashPartitions SH file and also the FlashPartitions .bat file "fastboot flash vendor [...]

Must I do this before, or after installing gamma os?

Kamins2017, 2024-07-03:
> Must I do this before, or after installing gamma os?

Before.

GMbizi, 2024-07-03:
> Well, I said
>
>
> So, my case is resolved! I said I edited the .sh file and no luck, but I tried editing the .bat directly and voilà, all worked like a charm. Thanks @TheGammaSqueeze for all your work

It's cool but I don't know if it's possible to solve the problem with an update? I don't want to lose my data and do all these manipulations

noriammx, 2024-07-07:
> I'm not sure why this happens on newer units, but the vendor_custom.img that ships with GammaOS is not compatible, even though this is based on stock firmware.
> 
> To resolve your issue, you will need to flash back to the stock firmware.
> 
> Then reflash GammaOS, however one change needs to be made in the FlashPartition script before flashing, and that is to remove the following line:
> 
> Remove this: `fastboot flash vendor flash/vendor_custom.img >>log.txt 2>&1`
> 
> This should resolve the issue for newer units.

Hello thank you very much.

I want to confirm that following these steps the WIFI worked for me.

I bought my console at the beginning of May.

Fox1304, 2024-07-10:
@TheGammaSqueeze is there a way to rollback to stock & then re-flash when all I have available is a Mac ? 
I'm on the same situation, and from what I understand reflashing to stock is only available on Windows ?

notanaijin, 2024-07-13:
I just bought my RG405M, its on its way. Is there a way to check if I need to remove the line from the FlashPartition script?  or is it safe to assume I should do it anyway? @TheGammaSqueeze 

I'd rather save myself time flashing, checking and reflashing if it fails.

Mr-BrightsideG, 2024-08-02:
> I'm not sure why this happens on newer units, but the vendor_custom.img that ships with GammaOS is not compatible, even though this is based on stock firmware.
>
> To resolve your issue, you will need [...]

Hello, I have always been aware of what was being done in this forum, a couple of days ago I followed the steps and I can confirm that it worked the first time and without problems on my rg405v, I decided to do multiple tests before commenting here, So anyone who has questions can do so with complete confidence, I am happy to know that a solution was found.

Mr-BrightsideG, 2024-08-02:
> I'm not sure why this happens on newer units, but the vendor_custom.img that ships with GammaOS is not compatible, even though this is based on stock firmware.
>
> To resolve your issue, you will need [...]

Hello, I have always been aware of what was said in this forum, I saw that they were commenting for some time about a possible solution to the wifi problem in gammaos, a couple of days ago I decided to follow the steps and I can confirm that it worked the first one and without problems on my rg405v. I preferred to do several tests before commenting here and it only remains to say that for those who have doubts, know that you can do it with complete confidence, I am happy to know that a solution was found.

francisgro, 2024-08-14:
Hello, many thanks for providing this workaround. I have a question: I am about to flash GammaOS onto a brand new rg505. I have only a Mac at hand so no way of going back to stock firmware for me afaik. Is it save to remove this line 
fastboot flash vendor flash/vendor_custom.img  >>log.txt 2>&1
on the first attempt? Are there any features missing if vendor_custom.img is not flashed?

SigmaMX, 2024-09-04:
Hey guys! I'd like to request help on the unbricking process. After the upgradedownload app starts, I press the cogwheel, load the RG405V file, it loads up, I press PLAY, I Plug the console, I press power+vol down+home, it shuts off, the app starts, it closes suddenly and the console turns off. As i turn it on with power+vol down the abernic screen shows as in the video but boots Gamma again! Thanks in advance for any help

notanaijin, 2024-10-08:
That's what I wanted to know, but it looks like anyone who knows anything isn't active here anymore unfortunately. I received mine and I was lucky and didn't have the issue, so the normal method worked totally fine. Maybe you'll have the same luck hopefully

ovilanova, 2024-10-08:
My personal story short:
I boguht a new RG505 last June (2024). I tried to install GammaOS but failed and (almost) bricked my device, probably by not following the steps 100%...
Then I used the Unbricker tool and successfully flashed stock OS.
Next I tried one more time to flash GammaOS (removing the vendor_custom.img line from FlashPartitions.bat, to ensure working wifi)
And everything work flawlessly! (tested some GB, GBA and SNES games so far...)

francisgro, 2024-10-09:
Hi,

I flashed GammaOS without vendor_custom.img directly on a brand new RG505 (bought in August, 2024). It works without any issues.

a4michel, 2024-10-16:
> is there any risk to just deleting the line `fastboot flash vendor flash/vendor_custom.img >>log.txt 2>&1` before starting the first install? I only have a Mac so having to source a Windows machine to use the unbricker will be...challenging...

Same for me, any news on this idea (removing the line entirely to be safe)?

mikkelnl, 2024-10-16:
Same for me, any news on this idea (removing the line entirely to be safe)?

charlesrosswood, 2024-10-16:
Well I know it's not of much use, but I just flashed Gamma OS to my RG405M (using MacOS) and it worked flawlessly without any modifications to `FlashPartitions.sh`.

eroldru, 2024-11-25:
Worked for me after the first reinstall :) Thanks to everyone for the tip

finalhopestudios, 2024-12-07:
Deleted the vendor_custom.img line from the batch file and installed on a new RG405M. Works fine.

wiseyard, 2025-04-07:
Just received my 405V, flashed SammaOs (no wifi) then used the unbrick tool to get back to factory os and flash again GammOs and still no wifi :( 

Any clue?

wiseyard, 2025-04-07:
removing fastboot flash vendor flash/vendor_custom.img  >>log.txt 2>&1 fixed the problem!

chandeeps, 2025-05-08:
> Hi,
> 
> I flashed GammaOS without vendor_custom.img directly on a brand new RG505 (bought in August, 2024). It works without any issues.

Thank you, i also removed vendor_custom.img as specified in above thread and it worked. I brought device in April 2025