TheGammaSqueeze/GammaOS #157: RG505 WiFi Issue GammaOS v1.5.1

mrkushaljoshi, 2024-04-04:
No issues with Wi-Fi on Stock Anbernic OS. 
Issue on GammaOS v1.5.1.
Reverted back to Stock Anbernic OS.

TheGammaSqueeze, 2024-05-24:
I'm not sure why this happens on newer units, but the vendor_custom.img that ships with GammaOS is not compatible, even though this is based on stock firmware.

To resolve your issue, you will need to flash back to the stock firmware.

Then reflash GammaOS, however one change needs to be made in the FlashPartition script before flashing, and that is to remove the following line:

Remove this:
`fastboot flash vendor flash/vendor_custom.img  >>log.txt 2>&1`

This should resolve the issue for newer units.

TEOPEMA, 2026-02-25:
> > Я не уверен, почему это происходит на более новых устройствах, но файл vendor_custom.img, поставляемый с GammaOS, несовместим, несмотря на то, что он основан на стандартной прошивке.
> > Для решения вашей проблемы вам потребуется вернуться к заводской прошивке.
> > Затем перепрошейте GammaOS, однако перед прошивкой необходимо внести одно изменение в скрипт FlashPartition: удалить следующую строку.
> > Удалите это:`fastboot flash vendor flash/vendor_custom.img >>log.txt 2>&1`
> > Это должно решить проблему для более новых устройств.
> 
> Как удалить строку? Пожалуйста, покажите мне способ...

Измените расширение файла из .bat в .txt 
Откройте с помощью "блокнот" удалите строку, сохраните и верните обратно в .bat