TheGammaSqueeze/GammaOS #160: There is no EraseUserData.bat file in the folder

andygreen6, 2024-04-08:
Do you have a direct file download for this?
Thanks in advance.

dfreivald, 2024-04-08:
Is there a file named "EraseUserData" ?

andygreen6, 2024-04-08:
EraseUserData.sh only. No EraseUserData.bat file. Im using windows btw.

dfreivald, 2024-04-08:
Make sure "Hide extensions for known file types" is unchecked.
If you really don't have it, make sure your antivirus isn't deleting it.

andygreen6, 2024-04-08:
Thanks for the tip!
I tried to redownload your updated file and it worked.

dfreivald, 2024-04-08:
Awesome, although all the credit goes to @TheGammaSqueeze 

This is a common issue, so here's the contents of the file in case others also need it.

```
@echo off
echo Ignore any errors.
echo.
fastboot erase metadata
fastboot format metadata
fastboot erase cache
fastboot format cache
fastboot erase userdata
fastboot format userdata
fastboot -w

echo.
echo Ignore any errors. /userdata and /metadata wiped. Factory reset complete. Closing this window in 60 seconds.
timeout /t 60

```

andygreen6, 2024-04-08:
Thanks again for the help. And ofcourse, many thanks  @TheGammaSqueeze
<https://github.com/TheGammaSqueeze>!!

closed 2024-04-18 by TheGammaSqueeze