TheGammaSqueeze/GammaOS #161: RG405V Can't Unbrick

Geeron510, 2024-04-10:
Recently updated to GammaOS on my RG405V (recently purchased from Anbernic's website) and had issues with connecting to WiFi.
The device would constantly just keep searching for networks and nothing would pop up. After looking into it I saw that my best option was to unbrick the device and go back to Stock OS and try to go back to Gamma OS after that.

However I'm unable to go back to Stock OS, I followed the steps correctly trying to unbrick the device, as well as used the proper firmware making sure it was for the RG405V but when it comes to connecting my device after setting the firmware for the UpgradeDownload, once I press and hold the HOME + VOLUME DOWN + POWER BUTTON my device just reboots, and I get a Failed [PS2264] Open Port Fail.
I've tried using a different USB to USB-C cable, as well as a different port but my device does the same thing. Can anyone shed light or help me out with this situation?

Geeron510, 2024-04-10:
[image]

TheGammaSqueeze, 2024-04-26:
This is usually down to a bad port or USB cable.

Try changing those, failing that then try another computer.