TheGammaSqueeze/GammaOS #164: No wifi after reflashing GammaOS RG 405V

monteleonis, 2024-04-18:
I saw the videos detailing how GammaOS would be the go to software to use for the 405V so after flashing the firmware over to the device and starting it up I came across an issue where there were no wifi networks available to connect to during the set up process for GammaOS. Thinking I downloaded something wrong like the wrong firmware, I unbricked the device, verified that the wifi works just fine on the stock OS, and re-flashed over the proper software. Again when on Set Up it's infinitely searching for WIFI networks. I can't download any emulators off Gamma or utilize any of the online apps, so it was kind of a big bust. 
Is there a work around for this? I've tried turning off the private DRN option with no luck, and tried using the command prompt to do it too but still nothing. I just got this thing from Anbernic not 3 days ago and I was hoping to use this software. Otherwise I'm going to have to go back to Retroarch.

TheGammaSqueeze, 2024-04-18:
https://github.com/TheGammaSqueeze/GammaOS/issues/89#issuecomment-1773807115

closed 2024-04-18 by TheGammaSqueeze

monteleonis, 2024-04-18:
> [#89 (comment)](https://github.com/TheGammaSqueeze/GammaOS/issues/89#issuecomment-1773807115)

I didn't download the 405M firmware though I specifically downloaded the 405V so this doesn't make any sense to me unless the download links are just straight up incorrect.

hpcraftwerk, 2024-04-26:
I am having this same issue, flashed with Gamma 405v and Gamma 405v Lite, neither one will see Wifi.

Used the Unbricker with Firmware 1.23.pac to restore and try each time.

Don4vi, 2024-05-26:
I have the same problem. I downloaded the RG405V firmware for the RG405V and the same thing.

TheGammaSqueeze, 2024-05-26:
https://github.com/TheGammaSqueeze/GammaOS/issues/167#issuecomment-2130061521

See the above

Don4vi, 2024-05-26:
Thank you very much for the reply. I have tried the unbricker method twice but the same problem continues. With the original OS I can scan Wi-Fi networks, but not with GammaOS. I've had the 405V for a week. I don't know if this is related to the problem of being too new.

geedeck, 2025-03-05:
So I'm reflashing to stock

And it doesn't see Wifi at all even on stock