TheGammaSqueeze/GammaOS #165: Possible to full restore partitions to sd card

informadortech, 2024-04-18:
Hi,

not really an issue, but I would like to know if its possible to make a full backup using recovery or other tool, that includes all partitions. So that I can restore to another identical console. If so, can you explain how? Tried with the recovery but I don't see that option.

TheGammaSqueeze, 2024-04-26:
You can do this via recovery and using adb shell.

First go into recovery.
Use the command adb root
Then you can do adb pull.

e.g 
adb pull /dev/block/by-name/boot_a boot_a.img

informadortech, 2024-04-26:
> You can do this via recovery and using adb shell.
> 
> First go into recovery. Use the command adb root Then you can do adb pull.
> 
> e.g adb pull /dev/block/by-name/boot_a boot_a.img

Great, and to restore?

TheGammaSqueeze, 2024-04-26:
I am not officially supporting what you are asking for, so do this at your own risk.

adb push boot_a /dev/block/by-name/boot_a

Though you are better off using fastboot for this.
fastboot flash boot_a boot_a.img

informadortech, 2024-04-26:
Will try and I can provide feedback later

informadortech, 2024-04-26:
1 -I can only pull without the .img extension. do I need to than just rename the file and add the extension?
2 - Do I need to do this for every single file present on the by-name folder? Or just the one's you also use on your Flashpartitions script?