TheGammaSqueeze/GammaOS #167: 405v Wifi Not working with Full or Lite

hpcraftwerk, 2024-04-26:
I saw that the other issue was closed with the reason being that the 405m firmware was used. I have attempted to flash with the 405v full and lite multiple times, and using the unbricker with 405v firmware 1.23 to return it each time. I have had no luck getting wifi to work on this machine.

Anyone have any other ideas, or just stuck with stock for now? Thanks again

TheGammaSqueeze, 2024-04-26:
What is specifically happening? Can you not see wifi networks? Are you able to grab a logcat log to share here?

hpcraftwerk, 2024-04-26:
Hi 

I boots into Gamma OS and Gamma Lite with no issues, but when it searches for Wifi Network is just does not find any and will hang indefinitely looking.  Stock OS is able to connect right away to Home Network.

I would be happy to grab you a log if you can provide steps. I am actually in the process of retrying Gamma Lite again right now.

hpcraftwerk, 2024-04-26:
Fresh install of Lite and no go, just hangs on Searching for Network. Appreciate your efforts!

rikmcastro, 2024-05-14:
I'm facing the same issue. Gamma OS boots but it keeps looking for Wifi Network indefinitely. Stock OS is able to connect to my home network, though. I'd appreciate the help, folks!

IAmGrover, 2024-05-19:
I can add that this is a reoccurring issue for myself as well. I've seen the other tickets stating that is could be an issue with installing the wrong device firmware and when that's not the case, flashing GammaOS, unbricking the device, and reflashing GammaOS should resolve the issue. 

This process has not resolved the issue for me, and I have confirmed this on both GammaOS and GammaLite. Wifi runs fine on Stock OS, but when installing Gamma the wifi keeps turning itself off, never even searches for available networks, and sometimes outright stops responding when opening the network settings and shuts you out. I'm hoping there is a solution to this, because you can't download new emulators without access to the app store.

Quinntuas, 2024-06-18:
This issue happened to me as well. Tried to flash back and forth several times and nothing changed. I did delete the script in FlashPartition.sh

Kamins2017, 2024-06-21:
Also looking for a fix for my 405v. Tried changing the script file but still no wifi.

BlinkersFr31, 2024-06-24:
Same problem for me.

mzapatero, 2024-06-25:
Same problem for me, I just got a brand new 405v and no wifi available.

lokize, 2024-06-27:
Same problem here, any way to fix without reinstalling stock os ?

luigira, 2024-06-29:
Same, I just flashed GammaOS and no wifi, stock was working fine. Can I suggest a warning on readme?

TheGammaSqueeze, 2024-06-30:
Please redirect all conversion to this issue
https://github.com/TheGammaSqueeze/GammaOS/issues/155#issuecomment-2189190537

closed 2024-06-30 by TheGammaSqueeze