TheGammaSqueeze/GammaOS #171: Upgrade to Magisk 27

DieHardMan5x, 2024-05-11:
Just flashed latest v1.5.1 on my RG405V, Magix wants an upgrade from v26.1 to v27.0.

[image]

But the diect installation fails with an error:
[magisk_install_log_2024-05-11T17.10.54.txt](https://github.com/TheGammaSqueeze/GammaOS/files/15283652/magisk_install_log_2024-05-11T17.10.54.txt)

DieHardMan5x, 2024-05-11:
**Gamma:**
Ah it'll never work by direct upgrade within the app.

You'll need to patch the boot.img manually

You should be able to get rid of that error message by copying the boot.img from the GammaOS folder you flashed on your 405m storage, then use magisk to patch the image 

Don't even need to flash it, just reboot. The error will be gone

closed 2024-05-11 by DieHardMan5x

noexpo, 2024-06-17:
I'm experiencing this same issue on a 405m.

Magisk is at 26.1, Zygisk: Yes, Ramdisk: Yes.

I flashed GammaOS Lite 1.5.1

Is the appropriate image file: boot_custom.img?

Schnitzelbrain, 2024-07-01:
> **Gamma:** Ah it'll never work by direct upgrade within the app.
>
> You'll need to patch the boot.img manually
>
> You should be able to get rid of that error message by copying the boot.img from the [...]

I tried that using the boot_custom.img. It patches and then saves the file. But nothing changes regarding the problem after reboot.

However at XDA they mention this problem as some seem not to allow direkt flashing from within android. 
[XDA](https://xdaforums.com/t/cannot-update-from-magisk-26-1-to-27-0.4658327/)

They also mention a fix 
`Did that, flashed through twrp and got a bootloop.
However, it succeeds by flashing magisk27.apk as a zip file through twrp`

Can this topic be reopened until the issue is solved