TheGammaSqueeze/GammaOS #174: Stuck flash vbmeta_a

Noel1987, 2024-05-19:
Hello 
Sorry for my Bad english 

I want to flash Gamma os

Bootloader ist unlocked 

Flash via Script Starts and stuck on 

Fastboot flash vbmeta_a flash/vbmeta_custom.img

All fastboot exe deleted and i use ADB fastboot++ tool 

What is the problem and what can i do?

Noel1987, 2024-05-26:
I used a complete new Windows PC 

[image]

DRCRecoveryData, 2024-07-19:
> All drivers are installed? Try another USB cable? Try another USB port?

Hi,

How to flash addy yan gsi to rg353v, when i flash to system after flash gsi rom is stuck on logo boot