TheGammaSqueeze/GammaOS #180: Could a new rg556 be requested from Anbernic?

slrgt, 2024-06-03:
Any chance you could request another rg556 so you could continue work on GammaOS on the rg556? Its an amazing device and would be great to have such a great OS on it! Especially so the gulli kit hall sticks could have the software fixed

Adrenalyzer, 2024-06-06:
I'd like to see the same on the RG Cube as it looks like a very capable successor to the 405M. The T820 vs T618 is a huge performance increase and the form factor is impressive. 

If Squeeze is interested in porting the FW we could all donate some paypal and he could give it a spin?

Thank you.

david79111, 2024-06-07:
Or somebody asks anbernic to send him a device or two as it is beneficial for them as well to have someone improve their products.

Adrenalyzer, 2024-06-09:
He had a RG556 that was bricked during development:

https://i.ibb.co/cgjrwVT/Screenshot-20240314-160938-Discord.jpg

It would probably be better for us all to chip in and buy him a UFS reader/writer so resurrecting the machine is possible and that could lead to cracking the T820 security which paves the way for CFW on the Cube as well.

Either that or Anbernic release the unbricker and it massively speeds up development otherwise its a case of use the hardware as it is and don't expect any CFW any time soon as JELOS won't be at the tea party either.

david79111, 2024-06-10:
Did someone start a donation for this ufs reader/writer where I can join?
Did anyone ask anbernic to release an unbricker yet?

david79111, 2024-06-10:
I emailed anbernic for help on this. Might not do much but at least I gave it a shot.

TheGammaSqueeze, 2024-06-15:
Not required. I have developed a working unbricker.

closed 2024-06-15 by TheGammaSqueeze

slrgt, 2024-06-18:
> Not required. I have developed a working unbricker.

Thats great! does that mean GammaOS on the 556 will be developed?

david79111, 2024-06-19:
It seems so. You can go to his page and look at his activity. There you can see that he is working on unlocking the bootloader for t820 specifically rg556 and rg cube. He cant start making a or rather flashing any os/rom before he or someone helps him unlock the bootloader though.

david79111, 2024-06-19:
In case this is any help for unlocking the bootloader

https://github.com/patrislav1/unisoc-unlock

shiftyscales, 2024-06-25:
@TheGammaSqueeze - Apologies if this is not the best place to ask- but would that tool work for the RG Cube as well? I am currently in contact with Anbernic Support, but they wish for me to ship the device back, and do not seem cooperative with the premise of attempting repair first.

As far as I can tell the hardware itself should be functional- but the unit bootloops. By holding Home / Volume Down / Power after the device has run out of power (to exit the bootloop) I am able to enter a recovery menu that shows the Android logo, but I can not find any adb drivers or any way to meaningfully interface with the device while it is in that mode. As far as I can tell there isn't a recovery partition- or at least any way I'm aware of to interface with it from what I have so far.

If there is any way I could repair it myself, I would prefer that to returning it and waiting around a month or longer round trip for them to send back a replacement unit if they do at all.

TheGammaSqueeze, 2024-06-25:
Unbrickers are available here:

556: https://drive.google.com/file/d/1hYuUPBTHAghZXXVt5urnAhJP-_vofg3K/view?usp=sharing
Cube: https://drive.google.com/file/d/1DBl9ZwvyqUdjMmetA2ekYWKS1uUcmI0b/view?usp=sharing

Same instructions apply as the 405/505 unbricker, just use these files instead.

shiftyscales, 2024-06-26:
Hmm- as mentioned- my device is bootlooping- I've not made any modification to my device nor had I unlocked the bootloader or anything else of the sort prior to this issue occuring- the device had locked up during its first use, and has continued to bootloop continually since after bootlooping the entirety of the first day I got it, the battery ran dry so I could experiment from a cold boot as the hotkeys, and power button were entirely unresponsive while it was boot looping.

I've noticed that when plugging in the USB cable it very briefly shows the download progressing- but immediately halts and claims the USB has been disconnected as the device enters the battery charge screen.

I either can't seem to get the timing down- or there's some other factor I'm not considering for why this isn't working. Any tips, @TheGammaSqueeze?

Have you seen one of these devices bootloop (load the Anbernic logo, fail to load into Android, reboot itself and repeat indefinitely until out of charge)?

TheGammaSqueeze, 2024-06-26:
> Hmm- as mentioned- my device is bootlooping- I've not made any modification to my device nor had I unlocked the bootloader or anything else of the sort prior to this issue occuring- the device had [...]

Boot loops in this case are probably caused by bad eMMC/UFS, so it's very well could be a hardware issue.
Try running the unbricker software, press play, then try holding down the Home/Back button during the boot loop to force it into BROM mode (nothing will be displayed on screen). It should start unbricking.

shiftyscales, 2024-06-26:
OMG!! It worked!!! Thank you so much. Oh my gosh... You're the best.

UnknownH, 2024-10-05:
First of all thank you for your astounding job!

I didn't know where to post this exactly, but it seems to be highly related with the last few posts here, so I thought that maybe you could help me as well.
My RG556 was working fine on day one, then I left it on standby overnight, but the morning after the device wouldn't turn on (apparently, at least the screen wouldn't switch on), and trying to charge would do nothing, not even switching on the charging led, it felt as if it was dead... It seems like it bricked itself overnight, and I'm not exactly sure how or why because the only thing I did was to store some updated emulators, testing roms and an sdcard. I know that it's not completely broken because if I connect it to my computer and hold Home+Power(+Volume up? not sure if this last one is optional) then I'll hear Windows recognizing a new device.

I've tried to follow [your guide](https://github.com/TheGammaSqueeze/Bootloader_Unlock_Anbernic_T820/releases) on how to unbrick T820 devices, but from a (hypothetically) switched off device, if I press the Home button and connect the USB cable, then nothing happens. nor the application will detect it of course. I need to also press the power button for the application to start doing something, but then it always fails after less than a minute with "usb_send failed (0 / 8)". I can't see anything on the device's screen screen regardless of how I try to switch it on. Also the charging led will never turn on no matter what USB cable I use.

I presume that my device is in a worse bricked state than the one assumed in the guide? But I though maybe you or someone else found themselves in the same situation and found a way to unbrick it? Thank you in any case