TheGammaSqueeze/GammaOS #183: RG505 - flash of GammaOS v1.5.1 is stuck at vbmeta_a

fdimeglio, 2024-06-14:
Hello,

I finally decided to flash GammaOS on my Anbernic 505 :-)  I am on Linux and very used to flash everything by myself for any Android OS based device. 

I have unlocked the bootloader as required and verified that it was unlocked by looking at the message in fastboot mode.

Then I tried to run the provided FlashPartitions.sh script and I was immediately stuck at the vbmeta_a flash with no progression.

So I looked at the shell script and did by hand:

$ fastboot flash vbmeta_a flash/vbmeta_custom.img
Sending 'vbmeta_a' (1024 KB)                       OKAY [  0.028s]
Writing 'vbmeta_a'      

and I am stuck again.

What am I missing here ?

Thank you in advance for your response

nhhrky, 2024-06-15:
I'm having the saame issue

StefanNedelchev, 2024-06-19:
Are you sure that you are actually stuck? This process is said to take up to 10 minutes but in my case it took even longer, but after a while it completed. Also after completion you shouldn't forget to run the other script which is for deleting user data (which also takes ~10 minutes.

fdimeglio, 2024-06-20:
Yes I confirm it is totally stuck. I let the FlashPartitions.sh script run for more than 3 hours and the flash did not even started and was stuck at trying to flash vbmeta_a.

I actually looked at the partitions and I got:

ums512_1h10:/ $  ls -la /dev/block/by-name/
total 0
drwxr-xr-x 2 root root 1580 1970-01-01 00:00 .
drwxr-xr-x 6 root root 3220 2024-06-20 23:43 ..
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 avbmeta_rs_a -> /dev/block/mmcblk0p68
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 avbmeta_rs_b -> /dev/block/mmcblk0p69
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 boot_a -> /dev/block/mmcblk0p42
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 boot_b -> /dev/block/mmcblk0p43
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 cache -> /dev/block/mmcblk0p53
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 common_rs1_a -> /dev/block/mmcblk0p70
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 common_rs1_b -> /dev/block/mmcblk0p71
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 common_rs2_a -> /dev/block/mmcblk0p72
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 common_rs2_b -> /dev/block/mmcblk0p73
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 dtb_a -> /dev/block/mmcblk0p48
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 dtb_b -> /dev/block/mmcblk0p49
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 dtbo_a -> /dev/block/mmcblk0p50
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 dtbo_b -> /dev/block/mmcblk0p51
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 fbootlogo -> /dev/block/mmcblk0p12
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 gnssmodem_a -> /dev/block/mmcblk0p19
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 gnssmodem_b -> /dev/block/mmcblk0p20
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 hypervsior_a -> /dev/block/mmcblk0p40
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 hypervsior_b -> /dev/block/mmcblk0p41
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 init_boot_a -> /dev/block/mmcblk0p46
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 init_boot_b -> /dev/block/mmcblk0p47
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_agdsp_a -> /dev/block/mmcblk0p32
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_agdsp_b -> /dev/block/mmcblk0p33
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_cdsp_a -> /dev/block/mmcblk0p34
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_cdsp_b -> /dev/block/mmcblk0p35
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_deltanv_a -> /dev/block/mmcblk0p26
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_deltanv_b -> /dev/block/mmcblk0p27
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_fixnv1_a -> /dev/block/mmcblk0p13
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_fixnv1_b -> /dev/block/mmcblk0p15
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_fixnv2_a -> /dev/block/mmcblk0p14
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_fixnv2_b -> /dev/block/mmcblk0p16
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_gdsp_a -> /dev/block/mmcblk0p28
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_gdsp_b -> /dev/block/mmcblk0p29
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_ldsp_a -> /dev/block/mmcblk0p30
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_ldsp_b -> /dev/block/mmcblk0p31
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_modem_a -> /dev/block/mmcblk0p24
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_modem_b -> /dev/block/mmcblk0p25
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_runtimenv1 -> /dev/block/mmcblk0p17
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 l_runtimenv2 -> /dev/block/mmcblk0p18
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 logo -> /dev/block/mmcblk0p11
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 metadata -> /dev/block/mmcblk0p56
lrwxrwxrwx 1 root root   20 1970-01-01 00:00 misc -> /dev/block/mmcblk0p3
lrwxrwxrwx 1 root root   20 1970-01-01 00:00 miscdata -> /dev/block/mmcblk0p2
lrwxrwxrwx 1 root root   18 1970-01-01 00:00 mmcblk0 -> /dev/block/mmcblk0
lrwxrwxrwx 1 root root   23 1970-01-01 00:00 mmcblk0boot0 -> /dev/block/mmcblk0boot0
lrwxrwxrwx 1 root root   23 1970-01-01 00:00 mmcblk0boot1 -> /dev/block/mmcblk0boot1
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 persist -> /dev/block/mmcblk0p23
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 pm_sys_a -> /dev/block/mmcblk0p36
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 pm_sys_b -> /dev/block/mmcblk0p37
lrwxrwxrwx 1 root root   20 1970-01-01 00:00 prodnv -> /dev/block/mmcblk0p1
lrwxrwxrwx 1 root root   20 1970-01-01 00:00 sml_a -> /dev/block/mmcblk0p6
lrwxrwxrwx 1 root root   20 1970-01-01 00:00 sml_b -> /dev/block/mmcblk0p7
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 super -> /dev/block/mmcblk0p52
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 sysdumpdb -> /dev/block/mmcblk0p57
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 teecfg_a -> /dev/block/mmcblk0p38
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 teecfg_b -> /dev/block/mmcblk0p39
lrwxrwxrwx 1 root root   20 1970-01-01 00:00 trustos_a -> /dev/block/mmcblk0p4
lrwxrwxrwx 1 root root   20 1970-01-01 00:00 trustos_b -> /dev/block/mmcblk0p5
lrwxrwxrwx 1 root root   20 1970-01-01 00:00 uboot_a -> /dev/block/mmcblk0p8
lrwxrwxrwx 1 root root   20 1970-01-01 00:00 uboot_b -> /dev/block/mmcblk0p9
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 uboot_log -> /dev/block/mmcblk0p10
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 userdata -> /dev/block/mmcblk0p74
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 vbmeta_a -> /dev/block/mmcblk0p54
[... 1 lines cut ...]
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 vbmeta_odm_a -> /dev/block/mmcblk0p66
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 vbmeta_odm_b -> /dev/block/mmcblk0p67
[... cut ...]
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 wcnmodem_a -> /dev/block/mmcblk0p21
lrwxrwxrwx 1 root root   21 1970-01-01 00:00 wcnmodem_b -> /dev/block/mmcblk0p22

So all the partitions the script is trying to deal with are there.  Something is preventing their flash.  BTW, I don't know why there is so many partitions compared to a standard lineage firmware...

My bootloader also seems correctly unlocked as the screen is saying the following after rebooting to the booloader:

"fastboot mode
INFO: LOCK FLAG IS : UNLOCK!!!"

Any idea ? 

Thanks,

fdimeglio, 2024-06-21:
Oh man, found the culprit.  I was in Fastboot mode and not in fastbootd.  When in fasbootd I was able to run FlashPartitions.sh and EraseUserData.sh scripts without any issue :-)

So for folks who don't know how to get into fastbootd, when your device was on Anbernic firmware, is to boot into the recovery with:

$adb reboot recovery

then you will select "Enter fastboot" with the Volume Up/Down and then press on POWER for validating the selection.

You should see "fasbootd" in red at the top and then you can use the FlashPartitions.sh and EraseUserData.sh scripts .

That said, I really like the work done and this is mile away better than the awful Anbernic firmware.  What a step up.  Thank you so much for doing that.

Now, may I ask to have the same for the Ayaneo Pocket S ?? ;-)