TheGammaSqueeze/GammaOS #185: RG405V joysticks out of balance

ITh1erry, 2024-06-19:
Hello there 

I am currently using GammaOS for about a week or so. Immediately when I got my RG405V I installed GammaOS. First and foremost, I am very happy with it and kudos to you guys!! But alas, yesterday I began experiencing a weird, maybe isolated issue. I noticed it especially when playing Super Mario 64. When pushing the left joystick to the right I noticed Mario wasn’t going as fast as opposed to pushing it to the left. 

First thing I assumed was that my joysticks were just faulty. I then downloaded a GamePad testing app to test the responsiveness of the sticks. And I saw a value of 0.79 when pushing to the right. When I then tried out the right joystick it was registering only a 0.65 on most axises. I was getting anxious. 

Then the weird thing happened. I fully restarted and powered off my RG and the values were different these times. It seems they can vary every time I reboot. But both of the sticks keep having around a 0.82 when pushing to the right. 

Maybe I’m totally wrong, that this could be related to the OS. But I found a similar post on Reddit which underlined my conclusions, see here: 
https://www.reddit.com/r/ANBERNIC/s/bHgTycCph2

So my question is: do you have any ideas to resolve this issue. Should I go back to factory settings and re-install GammaOS? I was almost thinking of installing new sticks, but I think the sticks are fine since the console is just about one week old. 

Thanks in advance guys!

ITh1erry, 2024-06-19:
Update: I tried out the recommended fix mentioned in issue #154 , I flashed the DTBO from V1.5.1 and it seems slightly better. Left stick now going to about 0.88 when I push to the right. Right stick going to 0.94 when pushing to the right. Still, the values seem to fluctuate on every reboot. So it still seems not fully optimized. FYI, I’m using GammaOS lite btw

ITh1erry, 2024-06-19:
There really is something weird going on.. now the values just went to almost perfect 1.0’s in every direction , just by switching games? Not even on reboot. At the beginning of this playtest, when I first powered on, I was seeing values in the 0.74’s again on my right stick. 

I really love the GammaOS, but I’d really like to have a consistent 1.0 in every direction. Hope you guys can help me out with this.

Suspicion/ suggestion of something to maybe look into: maybe something got scrambled by the ability to adjust the sensitivity of the analogs/ the new calibration thing? Anyway, no need for me to adjust the sensitivity. I just keep it on ‘normal’

dexwerx, 2024-06-24:
Make sure they are centered when you reboot, but it  sounds exactly like a hardware problem. You can try new sticks or use the custom mapping csv to decrease the point at which it registers as full scale.

ITh1erry, 2024-06-24:
Hi @dexwerx , yeah; I’ve been looking into that file you linked in the other issue. You were talking about someone managed to fix some out of spec sticks with this custom mapping. But how exactly did they do this. If I run this rgp2xbox.sh through Termux, it just crashes. Also tried updating the .c script myself already and compiling the program to just be able to fix the balancing. But alas, also an error. You maybe have some more pointers on this?

dexwerx, 2024-06-25:
You just modify the csv mapping file, and then use the quick settings tile to custom.

adingess, 2024-06-25:
I am also having the issue with a 405-V. My right joystick glitches out from -1 to 0 to positive 1 randomly. Is there anyway I can recalibrate this? Is this software related? Or do you think it is a faulty joystick?  

Unfortunately I'm going through some medical issues right now so I am legally blind, I can work on a computer decent enough with magnification turn on, so a simple solution would be best if possible. Thank you for looking into this.

TheGammaSqueeze, 2024-06-25:
@adingess this is probably a hardware issue with the stick itself. Software calibration won't fix that type of issue.

TheGammaSqueeze, 2024-06-25:
The CSV file that dexwerx is referring to is a feature that isn't documented yet, but essentially it will translate the raw input value of the stick to a value of your choosing. Others have modified this in the past to get around issues e.g sticks not returning to the center. 

In the future, I plan to expose the feature better in the UI so that we can calibrate sticks directly on the device itself.

adingess, 2024-06-25:
I had a hunched it was hardware-related. Just thought I would check just to make sure it wasn't anything I did or the software. Thank you for confirming. Also thank you guys for the great OS.

ITh1erry, 2024-06-25:
@TheGammaSqueeze thanks for the reply! It would be very helpful if we could calibrate and tweak through the UI. So you think my issue will be fixed in doing so? You replied to adingess’ issue that this is hardware related. What do you think of my particular issue? 

Oh yeah, btw, something else happened today. I also tried magisk to try out building my own module with your custom mapping script, like you mentioned to someone in another issue. I got a pop-up that I should flash / reinstall magisk? When I then chose to reinstall it crashed and I needed to reboot. All of a sudden the analog custom mapping quick setting says just ‘Off’. Actually, it seems I’m seeing more constant values for my joysticks now :’). So I still do think some tweaking could fix mine.

dexwerx, 2024-06-27:
When you reboot - make sure you're not touching the sticks. The center calibration happens at boot.

ToCarlosWebos, 2024-07-29:
I'm having exactly the same issue, it's absolutely not a hardware related problem, if you just go into stock OS the left Joystick will work again just fine, gammaOS is still very faulty for the rg405v, it's not worth for now, maybe in a year

dexwerx, 2024-07-29:
One of the things Gamma set out to fix when he created GammaOS was to fix the horrendous stock analog calibration and cardinal snapping. Unfortunately good calibration seems to be revealing the poor QC, of newer units. Without good calibration in the OS Anbernic doesn't even have a way to check if the sticks are good, so I'm not entirely surprised.