TheGammaSqueeze/GammaOS #191: Battery issues after recalibration on rg405v

Dantalion13, 2024-06-30:
Hello, I bought a RG405v and immediately installed gammaos 1.5.1

The battery stays high and drops in a few minutes from ~71% to 0%

I tried to recalibrate it by charging it to 100% and waited 2h more but it stays the same, going to 0% rapidly. 

I tried to use an app to recalibrate, but got no more luck.

have you got any solution to provide ?

Thanks for your answers,

[image]

lucasgomeshora222, 2024-07-01:
I have the same problem, which app did you use to recalibrate the battery?

Dantalion13, 2024-07-01:
I used one called battery calibration pro on the play store, it had 2 different process to recalibrate, used the longest one, depleting the battery to 0, charging, using it to calibrate at 100%+2h of charge then going to 0% and charging at 100% but turned off.

Spoiler : it didn't work.

EmilianoTaS, 2024-07-02:
Did anyone find a solution yet?? Im having the same issue... Battery is draining in 2 hours without using the console

quaddy, 2024-07-26:
Did you use a fast charger? Please don't. Use a 5V max. 2A USB Charger and try to recalibrate again. I've done this three times just with the OS without an app. Discharge, Charge, Discharge, Charge..