TheGammaSqueeze/GammaOS #200: Battery problem RG405M + GammaOS

trafnico, 2024-08-27:
Hi
I've just installed GammaOS, it's wonderful.
As other people on internet, I have this problem with the battery, which have no more autonomy.
It goes  very fast to 0%.

I tried to going to 0% (with a PS2 emulator for example), plug in the charger (with the good cable), waiting for 100% and then unplug.
The problem is still here.

Do you have a idea? a guide to how reprogram the battery ? Do I have to play with power saving mode + no wifi ?

Any advice will be appreciated 

Thanks a lot :)

trafnico, 2024-08-27:
[image]

here is a picture of my last try :(

trafnico, 2024-08-27:
[IMG_5628.pdf](https://github.com/user-attachments/files/16760149/IMG_5628.pdf)

here is the picture of my last try

dexwerx, 2024-08-28:
Unlocking and installation wipes the battery calibration. It takes a couple cycles from empty to full to get this working correctly.
When it dies, try and turn it on a couple times before plugging it in. Then make sure you boot into Android, and let it charge to full.

trafnico, 2024-08-28:
Ok ok !What do you mean by « boot in android » ?Another question, i have seen on forum that i have to open the anbernic after full cycle 0 to 100, then disconnect and reconnect the battery. Is it necessary ?

dexwerx, 2024-08-28:
When you plug in many of these devices, they will load a minimal boot system to charge the battery. This works fine to charge the battery, but it often won't update the calibration data that is part of Android.

By boot into Android I mean - when the battery is fully drained, go ahead and plug in your charger. You need to then hold down the power button to fully turn on the device, and load into the OS fully. Then you can leave the device plugged in and charging.

If you do this a couple times, you won't need to unplug the battery.

trafnico, 2024-08-29:
You're the best, I will try it ! 
Thank you so muuuchhh !!!