TheGammaSqueeze/GammaOS #215: 405m Stuck on Anbernic screen at last reboot.

Cha01605, 2024-12-30:
Fresh 405m, went through the entire process on a windows machine. I habe done many lineage/android flashes so the instructions were pretty simple/basic. After the last batch file was ran, I unplugged the 405 and hit the power button for the last reboot. It booted to the Anbernic screen (where it is supposed to stall for a few minutes before you see the Gamma screen) and it has been stuck here for over an hour. On the left side of the screen it says:

Fastboot mode
INFO: LOCK FLAG IS : UNLOCK!!!

Cant shut it down or connect to it through my PC.

Any ideas? It is the complete Gamma install, not the lite, if that matters.

TheGammaSqueeze, 2024-12-31:
Hi, I suggest looking in your folder that you flashed the system from, and check the log file there for clues.

In the worst case scenario, try the unbricking instructions.

Cha01605, 2024-12-31:
Thanks for the reply, Squeeze. I ended up being impatient and ran the unbrick tool without any further investigation. Everything worked great and Im back at the stock OS with unlocked bootloader. 

Now im having trouble getting the 405 to connect to the PC. Any commands I throw into ADB and Fastboot, I just get "waiting for any device". I have turned on developer options and also usb debugging. I feel like it is something with the 405 because one thing I notice, when I plug in my 405, I do not get the propt asking if its ok to connect on the 405. I know you always get that message when connecting an android device with usb debugging turned on. But I DO have a message on the 405 saying "USB Debugging connected. Tap to turn off USB Debugging." So its acting like it is connected...

Cha01605, 2024-12-31:
Im an idiot...
I was forgetting the adb reboot fastboot at the cmd propt...

Ran through everything again and im in Gamma. 

Thanks!

closed 2024-12-31 by Cha01605